yaapu/FrskyTelemetryScript #242: No telemetry data on ETHOS 1.5.11 Tandem  X20s

fdapport, 2024-08-11:
TX: Tandem X20s
SO TX: Ethos 1.5.11 FLEX
RX : R9 Slim +   868Mhz
SO Rx: 1.3.1
Controller: Matek F405 WIN
SO Controller: Arduplane V.4.0.8

On Open TX all work, include receiving telemetry data. I have installed Yapuu LUA end I have the following Error:
ONLY GPS coordinate sensors is discovered:  See below 

[image]

If I run the LUA Telemetry Debug all the sensors are received:

[image]

If I set LUA Sensors in the model 

[image]

I have the following error:

[image]

Please somebody could help me to removed the error. Thanks, Franco

fdapport, 2024-08-11:
I have run again with the last Yaapu version 1.1.0, same error but in a different line number (#1689) 

[image]

This is the wrong line...

[image]

yaapu, 2024-08-29:
Hi, tested this with 1.5.14 and it seems to work, can you upgrade and recheck? perhaps it was a bug?

fdapport, 2024-08-29:
Ok,  I will update the radio first and reload the scripts and I will let you know if it is all ok.... Thanks for the check. Ciao., Franco

fdapport, 2024-09-10:
I have update the radio to 1.5.15 version, yaapu works!!! . Thanks, Franco

closed 2024-09-10 by fdapport