yaapu/FrskyTelemetryScript #245: No telemetry data on ETHOS 1.5.15 Tandem X20pro

bad-wm, 2024-08-30:
TX: Tandem X20pro
SO TX: Ethos 1.5.15 fcc
RX : elrs 2.4
SO Rx: 3.3.0
Controller: FLRC H7D MK1 H743
SO Controller: Arduplane V.4.5.5
The ethos 1.5.15 system follows the installation method in the wiki and uses the ELRS receiver to always display 'no telemetry'. The widget can display the set feedback data such as link quality,
[image]

bad-wm, 2024-08-30:
[image]
[image]
[image]

yaapu, 2024-09-10:
Hi did you enable arduplane custom CRSF telemetry by setting RC_OPTIONS += 256?, this looks like an ardupilot configuration issue

bad-wm, 2024-09-14:
Okay, thank you very much. I overlooked the issue with ardupilot. Also, I would like to ask if yaapu only supports ardupilot, and if it supports BF or INAV.

yaapu, 2024-09-16:
as of now it requires Ardupilot

bad-wm, 2024-09-19:
Okay, thank you very much for answering my question

closed 2024-09-19 by bad-wm