yaapu/FrskyTelemetryScript #247: FrSky x18 attempt to index a nil value

alexandru-g, 2024-09-08:
Installed latest version of lua script on a FrSky X18 and getting the following error:
`/scripts/yaaputelemetry/lib/resetlib.lua:57: attempt to index a nil value (local 'widget')`

After this error appears, the ELRS TX doesn't appear in the menu anymore and I cannot configure it. I have to remove the yaapu lua scripts, reboot Ethos, configure the ELRS TX and re-add the yaapu scripts.

yaapu, 2024-09-10:
Hi, which Ethos version?

alexandru-g, 2024-09-13:
@yaapu 1.5.13

yaapu, 2024-09-13:
mhmm, I do not own that radio but on the simulator seems to work all right.
Waht would the exacts teps be to reproduce it?
Which module?
Does the elrs lua script work all right?
We might need to ask for help from somebody wit hthe same radio + tx combo :-(

alexandru-g, 2024-09-13:
RadioMaster Ranger Nano 2.4 is the ELRS module.

It works fine-ish. The entire radio lags a bit when I have the widget added (UI is not fluid, even when not in widget).
But I have to pause the widget so I can configure the TX and RX. Otherwise, when I press configure, nothing appears in the module list.

BRSS73, 2024-09-24:
Same here on a X20RS with Ethos 1.5.15
For reference: https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/4542

yaapu, 2024-09-25:
Did you pause my widget while working on the elrs config script?

BRSS73, 2024-09-26:
No, I wasn't aware of this possibility.

I found it quickly after your comment.

In the meantime, the ELRS script seems to have been modified by the Ethos team to handle concurrent access to the serial buffer, but maybe I'm wrong on that particular point.

alexandru-g, 2024-09-27:
> Did you pause my widget while working on the elrs config script?

Yes, that is the only way to enter the ELRS config script. If Yaapu is not paused, the TX/RX doesn't appear in the list.