yaapu/FrskyTelemetryScript #248: Telemetry script is freezing on TX16S running Crossfire and EdgeTX "Centurion" v2.10.4

marcdornan, 2024-09-12:
I am getting a very reliable script freeze after a few minutes of flight. The power button also freezes although thankfully the RC channels still work. Latest release of Edge Tx Centurion. I will post on EdgeTX github as well. This is not happening when I use SPort telemetry with Dragonlink (Mavlink) so I think this is related to CRSF.

marcdornan, 2024-09-17:
Update -- it seems Crossfire was giving very low telemetry rates (.3 hz) and this was causing the issue, especially after arming when there was more telemetry traffic. Once I updated and rebound I do not seem to be having the LUA script freezing.

yaapu, 2024-09-17:
hi, try to increase the telemetry to RC ratio

yaapu, 2024-09-18:
ok, I'm closing it

closed 2024-09-18 by yaapu