yaapu/FrskyTelemetryScript #251: No Telemetry data on TBS with Mavlink and CRSF

patrickpoirier51, 2024-09-19:
I can make mavlink over TBS working but I cannot get YAAPU to read the telemetry.... - EXCEPT for GPS…?!?

I see 18 values on the TELEMETRY screen including Hdg , Alt , Sats , Ptch , Roll, Yaw
YAAPU debug dont see a thing. And CRSF support is yes on the config file.

RadioMaster TX16 - edgetx 2.10
Yaapu Telemetry 2.0.x

yaapu, 2024-09-20:
Hi, this is not supposed to work, you need native passthrough over CRSF which requires CRSF as rc protocol on ardupilot not CRSF "forged" on the TX from mavlink frames which is what you are using

yaapu, 2024-09-20:
we user CRSF as a transport protocol and inject quite a number of custom telemetry frames

closed 2024-09-20 by yaapu