yaapu/FrskyTelemetryScript #252: Yaapu telemetry screen information and msg are not correct

fdapport, 2024-09-21:
Radio Hardware: Frsky X20S
Ethos Version :1.5.15 FLEX, latest
Yaapu telemetry widget version 1.1.0, latest
Yaapu Mapping Widge version: 1.0.0 dev

I have some issues as showed in the pictures:
1) GPS coordinate in the yaapu telemetry screen are not correct (see pictures)
2) Travel distance is NOT correct (64.95km!!!)
3) Bad compass msg are NOT correct, on the FPV video screen I sow the correct compass information, it look like fine for the aircraft direction
4) Some time I see error in the LUA scripts

Please help, Thanks, Franco

Telemetry coordinates look OK, Yaapu Mapping showed OK
[image]

[image]

LUA Error
[image]

yaapu, 2024-09-23:
Hi,
use the latest version of the widget for ethos 1.5.15 broke some GPS lua access.
Travel is wrong because it's calculated from wrong GPS coordinates, should be "right" with the new fixed version of the widget.
The main.lua issue has never been reported and I don't see it in my testing, try the new version and we'll see

yaapu, 2024-09-23:
Bad Compass Health has nothing to do with the widget, ardupilot is sending that message. You don't see it on the OSD because is related to you vehicle health status bits as reported by mavlink.

fdapport, 2024-09-23:
Thanks for the new version, just downloaded it and it works (correct coordinates) !!--.. for the bad Health you are right, I have the same msg in the telemetry from ArduPilot, no idea how I can fix...

yaapu, 2024-09-23:
Hi, could be a lot of things :-)
double check compass alignment configuration, calibration otmotor interference, best is asking in un ardupilot thread or discord