yaapu/FrskyTelemetryScript #254: White screen on the Yaapu telemetry MAP screen

fdapport, 2024-09-24:
I have some questions about yaapu telemetry screen:
1) EFF mAh (HSP/Curr - Efficency). When the model in the ground the values goes from 0 to 3000, than back to 0, it is correct?
2) I have set a additional screen to show the MAP but I have a white  screen as showed in the picture. I have set as well EthosMappingWidget and it works great!! Is it a error or I have make some error in the configuration?
3) How I set the wav language to Italian, in OpenTX I was able to set it. I have seen that we have in the SD the Italian WAv but I didn’t found where I can set it.

Thanks for the help,
Franco

[image]

[image]

fdapport, 2024-09-25:
Ok, great, Thanks for the fast reply.  
I will remove the EthosMappingWidget LUA script. I will keep the maps files.  
What about the set of the Italian Lanquage? 

Thanks again for the help. Ciao, Franco

yaapu, 2024-09-25:
Italian??
ahh I realized I'm missing the menu option... 
I added it, you should download the latest version from github and it's there :-)

sorry about that, ciao.

Ale

fdapport, 2024-09-27:
Installed, works great, Thanks, Franco

closed 2024-09-27 by fdapport