yaapu/FrskyTelemetryScript #256: Meaning of "T" and "F" symbols

joe-stroup, 2024-10-01:
I'm using the 1.9.5 Horus version on a RadioMaster TX16S and Crossfire.

I can't seem to locate information that explains what the "T" and "F" symbols that sometime appear on the screen.

Is there a place that explains these symbols?

[image]

yaapu, 2024-10-01:
Hi, documentation is lagging behind, sorry about that.
T= terrain
F= fence

when terrain is available and healthy, altitude will be reported as HAT (height above terrain) in green whilst altitude above home in white, if terrain is not healthy T will flash and an audio warning is triggered

if a fence is active an F is displyed, if you "break" the fence a warning is issued and F will flash

yaapu, 2024-10-01:
...and please update your widget to latest

yaapu, 2024-10-01:
[image]

closed 2024-10-01 by yaapu