yaapu/FrskyTelemetryScript #257: How do i install is on ethos? X18

Asafsol, 2024-10-04:
I dont see any instructions

Asafsol, 2024-10-04:
after i move the main lua file to scripts folder, how do i setup it on the radio?

yaapu, 2024-10-07:
https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Ethos

closed 2024-10-07 by yaapu

Asafsol, 2024-10-07:
did that already
after i copied the files i go to define new widget but the yappu widget
does not appear as an option

Asaf

yaapu, 2024-10-08:
this is a configuration question, not an issue,please use the discuss blog so people can help you out
https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/2397

if you follow the provided setup instructions it will work all right, my widget is not just 1 file, it's a whole bunch, do as described and it will work