yaapu/FrskyTelemetryScript #259: Yaapu not working with Expresslrs Mavlink

iamwinq, 2024-10-15:
Hi Alex!

I have just updated the expresslrs to 3.5.1 and tried using the Mavlink RC, so basically it now uses Mavlink instead of CRSF protocol. I have followed the guide as listed here:
[https://www.expresslrs.org/software/mavlink/](url)

On my radio telemetry, I can fetch the telemetry data on the sensors page. However, it seems that the yaapu script does not recognize it.. any thoughts on this? I confirm that the yaapu script is working when using CRSF protocol, but only when using the Mavlink it does not work..

yaapu, 2024-10-16:
it is not supposed to work, you need native CRSF, check this
https://discuss.ardupilot.org/t/mavlink-over-tbs-crossfire-not-working/80942/49

closed 2024-10-16 by yaapu

menschel, 2024-12-21:
> it is not supposed to work, you need native CRSF, check this https://discuss.ardupilot.org/t/mavlink-over-tbs-crossfire-not-working/80942/49

That is not entirely correct, Sir.

As it seems Ardupilot already transmits most of the values used in the script as standard CRSF frames and when switching to Mavlink, ExpressLRS does translate Mavlink to these CRSF frames on a best effort basis.
https://github.com/ExpressLRS/ExpressLRS/blob/master/src/lib/MAVLink/MAVLink.cpp#L4
It therefore may be feasible to switch from the embedded FrSkyPassThrough values to CRSF provided values.

I already have a POC https://github.com/ExpressLRS/ExpressLRS/pull/3077 running for status text and one battery status value. I'm fighting with the format though.

I'll push a PR once I'm ready.