yaapu/FrskyTelemetryScript #260: How to switch to "map" mode

joe-stroup, 2024-10-16:
I've had great success using YAAPU on my RadioMaster TX16S for several years.

Today I updated my radio from the 1.95 version of Yaapu to the 2.xxx version.

I'm haven't found the release notes to find out all the differences - but so far it looks fine.

I also decided it was time to try the map display feature.  So I read the wiki on download (pre-caching) and processing maps, and loading them on to my radio's SD card.  No problem.

What I haven't been able to find anywhere is how to switch from the "flight instrument" view over to the "map" view.

I've found several older YouTube videos that go over the process of setting up a transmitter switch to zoom the map.  But I haven't been able to find anything that says how to display the map.

The older videos on YouTube seem to display a version of Yaapu that only has the maps - no flight instruments.  That makes me wonder if there's a separate widget displaying the maps - and I have to use one or the other - the "map" widget or the "flight instrument" widget.

The Yappu config menu has places for setting zoom levels (min, max, etc) and the map being used.  But I couldn't find anything yet that shows how to switch to display the map.

I would greatly appreciate any help or guidance.  Thanks!

[image]

joe-stroup, 2024-10-16:
I found a YouTube video in Hindi that I was able to use to guide me. 

https://youtu.be/wZNz1nOhB0k?si=BwOmzQKiDK2Y2jfb I setup a new telemetry page and set the second page up also use Yaapu.  The I went into the widget settings and selected page 5.  I'm not sure what all the other pages can do.

Now I still have to figure out how to properly set the zoom level.

[image]

yaapu, 2024-10-17:
no need for a dedicated page, simply add 2 channels in the mixer, I use 31 and 32 and assign 2 - 3pos-switches.
Then select them in the widget configuration as emulated page and wheel
[image]

Use the page one to toggle between messages, main panel and maps, and the second one to zoom in/out.
For zoom control, mid is do not change zoom, up is keep increasing, down is keep decreasing, you can also select how fast you want the zoom in/out feature.

joe-stroup, 2024-10-17:
Thank you for your kind reply and support.

Yes - I'd read that wiki you reference.

Unfortunately, it doesn't cover the necessary widget setup on OpenTX.

OpenTX Widgets can have up to 5 "main views."

And for each of these, there are eight screen types available for the Yaapu widget.

I couldn't find a reference for which of these eight screen types were used by Yaapu, and what they display.

I tested the first 5 screen types and it appears there are at least three: Instrument display,

Screen type "1" is the instrument display.

Screen type "2" displays MavLink Messages.

Screen type "3" is the instrument display - but with some odd additions.  It has arrows pointing to some of the instrument elements, and a flashing box around the entire screen.

Screen type "4" is the instrument display again.

Screen type "5" is the map display.

I haven't tried screen types 6 - 8 yet.

If Yaapu has more than 5 screen types, I'll have to pick and choose which to use on the 8 widget screens.

Can you please direct me to where the Yaapu screen types are identified?

I'm still not getting my map tiles to display.  I'll recheck my work before asking for help.

Thank you!

joe-stroup, 2024-10-17:
I finally got the map tiles to display.

Looking at the directory structure in the wiki, I discovered that the directory for the map tiles has changed with the upgrade to Yaapu 2.xxx.

There were a couple of helpful comments in the wiki notes about upgrading from 1.9.x about files that had to go in new directories with the new 2.xxx version.  I don't know how many directories have changed their location - but if it's not too much trouble, it would helpful to point them out in the upgrade notes.

Thanks!

yaapu, 2024-10-18:
Hi those notes point out all that's required to move around.
I added a really brief list of screen types here:
https://github.com/yaapu/FrskyTelemetryScript/wiki/Screen-layout#edgextopentx-color-radios-screen-types
It isn't much but at least it's there

closed 2024-10-18 by yaapu