yaapu/FrskyTelemetryScript #261: Rangefinder data on the Telemetry Plot screen does not appear

joe-stroup, 2024-10-25:
I recently added a CubePilot Here Flow optical flow sensor to my copter.

To help monitor the lidar rangefinder output, I thought I'd plot that on the Yaapu telemetry plotting page.

Unfortunately, no data appears:

[image]

Checking on Mission Planner - this data is available:

[image]

I tried other data - and most but not all of them worked for me.  I also repeated the "discover sensors" procedure - just in case the rangefinder data needed to be added.  I also tried plotting the Sonar data - just to make sure it wasn't using that data.

The rangefinder parameter names have changed a few revisions ago - so that it now easily supports multiple rangefinders.  The rangefinder parameter names now have a "number" following them to identify which rangefinder.  (They did the same thing with the camera parameter names.)  As a guess, I wonder if Yaapu uses the old parameter names without the number identifiers at the end.

If it's possible to get this data on the Yaapu telemetry plotting screen, I would appreciate it.

Is there something I need to do to make it appear?

yaapu, 2024-10-28:
Hi,
misison planner and passthrough telemetry are 2 different things, sensors are NOT involved, the only used sensor is GPS, all other telemetry data comes from a custom protocol.
No idea what your issue is

yaapu, 2024-10-28:
Hi,
I tried a simulated mission in SITL, I enabled rangefinder1, I can plot it with mavproxy and in my widget I see
[image]

So I guess the widget should work, again not sure what your issue might be :-(

joe-stroup, 2024-10-28:
Thank you for taking the time to test this.  I appreciate it!

If there is anything I can do to help your testing, please let me know.  I'd be happy to help.

About a year or so ago, there were changes to ArduPilot to more easily support multiple devices such as cameras and rangefinders.  Many parameter names were changed.

In Mission Planner both "sonarrange" and "rangefinder1" can be displayed - and show the same values.  In the BIN files, EKF3.RNG has this data.

Once again, I appreciate your help - and if I'll do my best to help.

yaapu, 2024-10-28:
I would say that if you can see it as mavlink "RANGEFINDER.distance" it shoudl work since both MAVLINK and frsky telemetry library use the same ardupilot source

yaapu, 2024-10-28:
the library uses the 
distance_cm_orient(270) so if your rangefinder is not facing down you might be getting 0?