yaapu/FrskyTelemetryScript #262: Unable to see telemetry from INAV on EdgeTX using Mapping widget

Painless360, 2024-11-28:
Thank you for looking at this. I am in the process of making a 'how-to' on the mapping widget and how to download and use the maps with the LUA script. 

I've hit a snag. INAV 7,1, telemetry coming back to the radio fine and sensor '1' is showing the lat/long in the telemetry screen in EdgeTX. I've deleted and redescovered the sensors but the Yappu LUA script is still showing NO TELEMETRY. 

INAV 7.1, EdgeTX 2.10.1, Yappu LUA script 2..0.2 dev (6cccb30), connection to FC via FrSky X16 receiver and telemetry coming back via Smartport from a Matek F405-MWN.

I've looked in the config menu and read the entire Wiki but can't see anything specific to INAV. Any advice welcome. 

Sorry to both you with such a simple one,, but stuff like this makes for a great PRO TIP for a how-to video! :D

Thanks again for the fantastic script. Happy flying!

yaapu, 2024-12-12:
Hi, you should use the Mapping widget not the Yaapu Telemetry Widget, the latter only supports Ardupilot.
Use https://github.com/yaapu/HorusMappingWidget instead which requires only a GPS sensor

yaapu, 2024-12-12:
@Painless360 please use the issue on the mapping widget page
https://github.com/yaapu/HorusMappingWidget/issues/61

closed 2024-12-12 by yaapu