yaapu/FrskyTelemetryScript #263: No telemetry with CSFR on Twin Xlite with TBS module

KPruett, 2024-12-26:
Hello.  

I'm trying to set up the yaapu telemetry script on my  Twin X-lite tx.

I'm using CSFR for RC to my flight controller,  and I have telemetry coming into the transmitter successfully, but get an error from the script,  and it's displaying "No Telemetry" in the widget. 

Im running the following: 

ETHOS 1.6
TBS crossfire nano module (latest firmware)
TBS crossfire nano Diversity RX (latest firmware)
Ardupilot 4.5.7 on a MatekSys F765 Wing. 
CSFR protocol for RC and telemetry.

ERROR MESSAGE:
RADIO: scripts/yaaputelemetry/lib/utils.lua:26:field 'popFrame' is not callable ( a nil value)

Any hints or suggestions would be helpful.  Thanks!

zebu14, 2025-01-30:
HEllo,
I haven't tried to downgrade but I may have the same problem on ETHOS 1.6.0 on FrSky X20RS.
"No telemetry" message on Yaapu 1.3.0 script, but ETHOS telemetry screen is showing data correctly.

Tx module BetaFPV Nano
Rx BetaFPV

External Tx type selected : ExpressLRS
Protocol selected on Rx Setup : CRSF

Any update on this issue ?

Thank you

yaapu, 2025-03-10:
Hi, check here please, should work
https://github.com/yaapu/FrskyTelemetryScript/issues/268