yaapu/FrskyTelemetryScript #264: Weird screen flashing with Ethos 1.6 and earlier 1.4

jaroszmm, 2024-12-31:
As in the attached clip. When powering up and down the radio the popup menu from switch warnings and the one asking if you want to switch off radio when reciever is connected makes yaapu script flash like this.
Nothing very annoying but I thought I'll share that. 

https://github.com/user-attachments/assets/c54e4064-7fd4-449d-9ddf-0869bf29a645

KPruett, 2025-01-08:
I am running ETHOS 1.5.19, and I experience the same flickering. Once the switch warning, or power down notifications are cleared, no more flashing.  It would appear that both the ETHOS software and the widget are competing to be the top display layer.

Tensho, 2025-01-31:
The same flickering behaviour is on ETHOS 1.6.0.

yaapu, 2025-03-10:
I removed an "optimization" which forced full screen redraw every 200ms, now redraw is at full speed and no more flickering but perhaps there will be a higher cpu usage, we'll see, please check this version and report if it works for you
https://github.com/yaapu/FrskyTelemetryScript/issues/268

closed 2025-03-10 by yaapu