yaapu/FrskyTelemetryScript #265: Jumper T15 Support

mathewcommons, 2025-01-11:
Tried to load yaapu telem onto my new controller jumper t15 and its not working.

yaapu, 2025-01-15:
Hi, T15 should work just fine, I tested on an actual radio, I suspect the issue is elsewhere, like configuration or proper ardupilot configuration, please describe your setup with edgetx version, ardupilot version and widget version

HishamGhosheh, 2025-03-27:
It works fine for me (testing it very recently)
Assuming you're using ELRS, did you enable CRSF from the yaapu config? You also need CRSF telemetry passthrough from the RC_OPTIONS parameter

[image]

HishamGhosheh, 2025-05-04:
Well, looks like there's some more complexity to this. 
It was working when I was on **EdgeTX "Centurion" v2.10.6**
When I updated to **EdgeTX "Jolly Mon" v2.11.0** it stopped working
The switches on the T15 used to indicate charge percentage while charging but that also stopped working. I suspect there's a bug in the EdgeTX firmware itself for this radio.
I have now reverted back to v2.10.6 and will stick with it.

### More details if that helps:

On EdgeTX "Jolly Mon" v2.11.0 the **screen 1** works and I can see the model tilt and mode change announcements but nothing on the Ardupilot messages (INF/DBG/...). Sometimes the screen would even freeze which was a bit worrying.
I also got `Emergency Mode` on the radio once. Not sure what caused it.

I verified my setup on EdgeTX "Jolly Mon" v2.11.0 and changed nothing on the Ardupilot side but it didn't work.

First screen sort of works. No Ardupilot messages.
[image]

Second screen never worked. Both screens complained about No Telemetry every now and then
[image]

I did have telemetry

[image]

[image]

[image]

I tried increasing telemetry rate and packet rate

[image]

[image]

[image]

[image]

yaapu, 2025-05-06:
Hi, edgetx 2.11 switched to Lua 5.3, my script is written for lua 5.2 and needs a full rewrite

closed 2025-06-05 by yaapu