yaapu/FrskyTelemetryScript #268: 'pop frame' not callible on Ethos 1.6

iamwinq, 2025-01-31:
[image]

I just updated my ethos fw to 1.6 on my Frsky Tandem X20S. Now I notice an error/warning icon on the top portion of the screen. It seems there is an error encountered calling this pop-frame?

iamwinq, 2025-01-31:
[image]

It only shows this screen, the script does not load unlike on the previous ethos fw

iamwinq, 2025-02-01:
I tried updating the yaapu script to 1.3 and it still shows the same error.

I also tried another setup, where it uses mav2pt, and internal module as telemetry source. It worked just fine... I guess the yaapu script only got issues with external module?

iamwinq, 2025-02-01:
rolled back to ETHOS 1.5.19 for the mean time, hope there will be a fix on yaapu soon!

iamwinq, 2025-03-04:
Hi @yaapu 

any update on this?

mikedkelly, 2025-03-06:
I also filed this in the Ethos github. Their reply was "In 1.6.2 telemetry frames are popped with new method (sensor)." I am filing a request for them to version control their lua API library.

yaapu, 2025-03-10:
Hi, could you please test this version?
[yaapu_widget_1.4.2_ethos_1.6.x.zip](https://github.com/user-attachments/files/19161698/yaapu_widget_1.4.2_ethos_1.6.x.zip)

iamwinq, 2025-03-10:
downloading :-) i will give feedback in a while

iamwinq, 2025-03-10:
> Hi, could you please test this version? [yaapu_widget_1.4.2_ethos_1.6.x.zip](https://github.com/user-attachments/files/19161698/yaapu_widget_1.4.2_ethos_1.6.x.zip)

I have just tested this with the latest Ethos 1.62 and it works now! :-) Thanks for your hard work @yaapu 

[image]
[image]
[image]
[image]
[image]

mikedkelly, 2025-03-10:
Thanks for continuing to support Ethos, Alex. I also tried internal and external RF and had no issues. There are no audio files if doing a clean install, but an overwrite install worked fine.

yaapu, 2025-03-10:
> Thanks for continuing to support Ethos, Alex. I also tried internal and external RF and had no issues. There are no audio files if doing a clean install, but an overwrite install worked fine.

The ethos team is giving me a hard time, pretty much they break the api at every major release, a real pain to support

mikedkelly, 2025-03-10:
Yes, agreed. I tried to advocate for some version control for the Lua API library, but it doesn't seem to be practical since they would need to keep multiple versions on the radio. I'm sure there is more they can do such as making new parameters non-required or creating a new API name if they refactor it entirely.

iamwinq, 2025-03-10:
that's fully understandable @yaapu . I just hope ETHOS wont break it again. 

@mikedkelly so for clean install, it isnt possible now, i guess this is just the patch :)

yaapu, 2025-03-10:
> that's fully understandable [@yaapu](https://github.com/yaapu) . I just hope ETHOS wont break it again.
> [@mikedkelly](https://github.com/mikedkelly) so for clean install, it isnt possible now, i guess this is just the patch :)

I updated the github repo, clean install as usual is now possible

yaapu, 2025-03-10:
> love you :) just for the sake of conversation, is this a frsky breaking things, or is this a new thing for new versions of lua in general? and also, is there some sort of donate now support me link? really love your work :) p.s. found the link :)

thanks a lot for your support!

iamwinq, 2025-03-11:
> > that's fully understandable [@yaapu](https://github.com/yaapu) . I just hope ETHOS wont break it again.
> > [@mikedkelly](https://github.com/mikedkelly) so for clean install, it isnt possible now, i guess this is just the patch :)
> 
> I updated the github repo, clean install as usual is now possible

Thank you again @yaapu  :-)

iamwinq, 2025-03-11:
> hm. it is still not behaving...it might be that I am missing some setup steps, but refresh is not very working, and I get some error now and then :/

flash to latest ETHOS (make sure to make a backup of your models), put the yaapu files in the sd card (Both the common and those folders specific for your radio (800x400 for  the x20s), overwrite the files. restart the radio and it should work fine. Oh, make sure to configure the yaapu widget settings as needed (enable CRSF support for ELRS and external module for the telem source, otherwise internal)

yaapu, 2025-03-12:
Hi, from the screenshot you posted you seem to have elrs version 1.0, it's old and does not support ardupilot, you need version 2.x.
After upgrading elrs you have to set a high telemetry rate, if you are on 333Hz @ 2.4Ghz use a 1:4 ratio, you can try 1:8 but not slower than that.
Remenber to enable ardupilot's custom CRSF telemetry with RC_OPTIONS += 256 (set bit 8)

iamwinq, 2025-03-13:
Set it up according to the documentation 

[image]

[https://github.com/yaapu/FrskyTelemetryScript/wiki/Passthrough-over-CRSF-and-ExpressLRS](url)

yaapu, 2025-03-13:
it's elrs firmware version should be >= 2.x

yaapu, 2025-03-14:
> hm. seems that changing telemetry rate to 1:2 1:4 or 1:8 even, works like a charm. updates instantly. thanx. however there is still this error:
> 
> ?:-1: attemtp to index a nil value both receiver and transmitter are on elrs 3.5.3 everything else seems to work perfectly :) thank you :)

attempt to index nil? when does it occur?

iamwinq, 2025-03-14:
I believe I have seen that error before as well, looks like file path issues, as you can see here 

[image]

yaapu, 2025-03-14:
yes, this means that the widget initialization code tryes a reset before the widget is fully initialized, once initialized it just works

iamwinq, 2025-03-14:
I usually ignore it, though I still wonder why the file paths are not the same. That "/./" Seems to mess it up 😂 good thing it does not seem to affect the operation of yaapu

yaapu, 2025-03-14:
> I usually ignore it, though I still wonder why the file paths are not the same. That "/./" Seems to mess it up 😂 good thing it does not seem to affect the operation of yaapu

that syntax is just a way to convert "relative" to absolute needed after 1.5.18 if I recall right

iamwinq, 2025-03-16:
oh, that's cool! thank you for clarifying this @yaapu :-)

closed 2025-06-05 by yaapu