yaapu/FrskyTelemetryScript #273: Syntax errors when installed on Radiomaster GX12

DavidRoote, 2025-04-15:
Hi all... Disclaimer first...  I am a complete novice when it comes to Lua scripts as well as Github issue logging so please forgive any mis-steps.

The issue I am having manifests as a syntax error on the assigned telemetry page when running either the Yaapu7 or Yaapu9 scrips on a GX12 radio.
In addition, a different syntax error appears when attempting to run the Yaapu config tool.

Error on telemetry page reads... "Syntax Error yaapu7.luac version mismatch in precompiled chunk" (or yaapu9.luac in the case of F2 and F4 versions)
Error when running the config tool reads... "Syntax error ?:0 attempt to call a null value"

For reference:
- I am running edgetx-gx12-v2.11.0-rc3 on the radio
- I have tried installing from both "latest release" as well as Ver 2.0.0_beta3 of the Yaapu script
-In case it is relevant, the other telemetry script on the radio (gplusl) runs fine (although its' telem page also gets masked by the yaapu syntax error when the yaapu script is loaded)
-The Yaapu Debug and Yaapu DebugCRSF tools appear to run okay (ie. no syntax errors)... just figures (5000:0 5001:0 5002:0 up to 5009:0... then 500A:0 through to 500D:0)

Please advise if I can be of further information.

Many thanks for all the creative efforts.

dmazan, 2025-04-22:
Try deleting the precompiled .luac files and replace them with the source .lua files from the /src directory.  That will let the radio recompile the source files and might fix the problem.

DavidRoote, 2025-04-23:
> Try deleting the precompiled .luac files and replace them with the source .lua files from the /src directory. That will let the radio recompile the source files and might fix the problem.

@dmazan ... Thanks very much for the guidance.. That did the trick.

For those who may follow, the files in the "generic" 128x64 black and white screen folder (Yaapu7.lua) worked on the GX12.. (The Yaapu9.lua appeared to cause the radio to hang)

Thanks again for all the creative (and engineering) efforts

closed 2025-04-23 by DavidRoote