yaapu/FrskyTelemetryScript #275: Missing console messages on Radiomaster GX12

dmazan, 2025-04-22:
Running Yaapu 2.1.0-dev on a new Radiomaster GX12 with EdgeTx 2.11.0, 2.4Ghz ELRS, Arducopter 4.5.7, CSRF passthrough telemetry, 500Hz (maximum available for the GX12), 1:2 telemetry ratio.  Everything works except no console messages.  Same vehicle, almost identical setup on a Radiomaster Pocket everything works including console messages.  The only differences on the Pocket are 1,000Hz packet rate and EdgeTx 2.10.1.

Is it possible I've missed a configuration option?  Any other thoughts before I try updating EdgeTx on the Pocket to see if that's the cause?

Yink059, 2025-05-01:
I got mine to work by adding local functions to replace both bit32.bxor and bit32.extract in the yaapu7.lua file

`local function bit32_extract(x, field, width)
  width = width or 1
  return(x >> field ) & ((1 << width) - 1)
end

local function bit32_bxor(a, b)
  local result = 0
  local bit = 1
  while a > 0 or b > 0 do
      local a_bit = a % 2
      local b_bit = b % 2
      if a_bit ~= b_bit then
          result = result + bit
      end
      a = math.floor(a / 2)
      b = math.floor(b / 2)
      bit = bit * 2
  end
  return result
end`

dmazan, 2025-05-03:
@Yink059 Thank you.  That does fix the problem, although it isn't clear why.  

Is there a breaking change in the implementation of the bit32 library in EdgeTx 2.11?  Or is the Yaapu script making an assumption regarding the behavior of those functions that no longer holds?  Or is it an issue specific to the implementation on the GX12?

What I'm getting at is that overriding those two functions seems like an expedient but temporary solution.  Should the long-term fix be in the EdgeTX Lua implementation or in the Yaapu script?

Yink059, 2025-05-03:
I think it's because 2.11 implemenented lua 5.3, which removed the bit32 library. older versions were using 5.2.

olliw42, 2025-05-14:
> I think it's because 2.11 implemenented lua 5.3, which deprecated the bit32 library.

to share, we use bit32 in the mLRS lua script, and it appears to just work fine on edgetx2.11 ... could suggest it is not bit32

Damirados, 2025-05-14:
> I got mine to work by adding local functions to replace both bit32.bxor and bit32.extract in the yaapu7.lua file
>
> local function bit32_extract(x, field, width) width = width or 1 return(x >> field ) [...]

Tried this fix but got syntax error popup after initiating autotune from althold mode.

yaapu, 2025-05-21:
Hi, you need a version compatible with edgetx 2.11, please try this one
https://github.com/yaapu/FrskyTelemetryScript/tree/pr/lua53

closed 2025-06-05 by yaapu