yaapu/FrskyTelemetryScript #277: No more status text messages after update to ETX 2.11.0/telemetry widget 2.0.x dev

dabit20, 2025-05-10:
I wanted to try the ELRS 3.5 MAVlink mode and did a round of firmware updating to get there. I installled a new ELRS TX module flashed with ELRS 3.5.4, updated EdgeTX on my TX16S to 2.11 and the telemetry widget to 2.0.x dev (1997425). 
I came from 1.9.6 and some older version of EdgeTX (2.8.something) which was working fine. 

ELRS MAVlink mode was not what I expected from it and the telemetry script did not work with it, so I went back to CRSF. Now I had telemetry again, except for the status text messages. The only message appearing is '00:00 DBG Yaapu Telemetry Widget 2.0.x dev (1997425)', nothing from Ardupilot (4.5.7). But for example the selected flight mode is displayed.

To make sure I did not kill something on the Ardupilot side I took another unmodified plane and re-installed the original/older ELRS TX module (running ELRS 3.3.something). No status text.

I tried downgrading the Yaapu telemetry widget to 1.9.6, but I was unsuccessful (widget not appearing in the model, unable to select the widget in the telemetry setup).

I did not yet downgrade EdgeTX.

Not sure what to do next.

jlpoltrack, 2025-05-10:
I also noticed this as well on mLRS, reverting to EdgeTx 2.10.5 resolved.

dabit20, 2025-05-11:
I can confirm that downgrading EdgeTX to 2.10.6 solves the issue with the status text messages.
Not yet sure if issue https://github.com/yaapu/FrskyTelemetryScript/issues/276 is resolved with the downgrade too.

tedsears, 2025-05-12:
Same issue for me.  Will try downgrading to EdgeTx 2.10.6

tedsears, 2025-05-12:
Downgrade to 2.10.6 fixed it for me too...

Amigache, 2025-05-14:
It seems like the rest of the functions are working correctly, except for the text statuses. I hope for a solution soon. Thanks!

dabit20, 2025-05-14:
Flew the entire evening yesterday, no issues with 2.10.6.

Amigache, 2025-05-14:
For me, version 2.11 is important to be able to use yaapu with DragonLink, without having to use a very old version of EdgeTX.

yaapu, 2025-05-20:
for color lcd radios this version should work on edgetx, please update the sound files as well
https://github.com/yaapu/FrskyTelemetryScript/tree/pr/lua53

Amigache, 2025-05-20:
@yaapu about my workbench, apparently it is working correctly for elrs and mlrs, I just need to try DL with Mav2PT, thanks for the effort!!

yaapu, 2025-05-20:
cool, the only issue that you might have is about audio notifications, I had to create new files with different names so just be sure to copy the new ones

Amigache, 2025-05-20:
> cool, the only issue that you might have is about audio notifications, I had to create new files with different names so just be sure to copy the new ones

Yes, it seems I copied everything. I just tested the script with DL and Mav2PT. EdgeTX finally includes the option to select s.port telemetry through the external module. Thanks again. I hope to be able to perform the flight tests over the weekend.

jlpoltrack, 2025-05-21:
Also resolves the issue for me with mLRS.  Tested on Jumper T15 with EdgeTX 2.11.1 

Thanks Yaapu!