yaapu/FrskyTelemetryScript #279: Performance Verbesserung in Ethos nötig

Voglberg, 2025-05-23:
Hallo zusammen, ich habe nun auf meiner FrSky X20R auch das Yaapu Widget am laufen. Aber ich muss sagen, flüssig läuft es noch nicht. Bitte um eine Performance Verbesserung.
Gruß und Danke.

yaapu, 2025-05-25:
new version is right here on github
https://github.com/yaapu/FrskyTelemetryScript/tree/master/ETHOS

yaapu, 2025-05-26:
Instructions are here
https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Ethos

If you need help don't use github issues but rather the discuss server
https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/2456

Voglberg, 2025-05-26:
Hallo, leider bin ich nicht der PC Freak. Bei mir lädt es immer nur den Web-Link zu der Seite runter, nicht die einzelne Datei.

Sorry für die Umstände.

Gruß Lothar

 

https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Ethos

https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/2456