yaapu/FrskyTelemetryScript #281: EdgeTX 2.11 issue (using latest yaapu script) Horus X10

nicesub, 2025-06-01:
Hi,

As per the title, I have updated my Horus X10 to the latest EdgeTX version as well as latest Yaapu script and I am having the following issues:

- No mavlink messages anymore listed on the screen
- After throwing my plane the script will crash and I will see the following error on my screen: "ERROR in refresh():?:0: attempt to perform 'n%0'"

Solutions tried:

- Reinstalling yaapu by removing all files related to yaapu from my SD and then placing new Yaapu files. The only folder that I did not delete is the one under the folder IMAGES as I have downloaded maps which would be a real pain to re-download again.

I am using the script with CRSF support enabled over ELRS (latest version of this as well).

Thank you for your support.

yaapu, 2025-06-05:
which version are you using because I had many reports of it working all right

nicesub, 2025-06-10:
Hi,

Thank you for the link! That did the trick. I used the link under the "Releases" section. I am no Github expert so I thought that must be the latest version as it has recent date on it as well.

closed 2025-06-24 by yaapu